RPi-Distro/vlc #101: VLC does not play some HLS streams, but OMXplayer does

eddsor, 2024-01-06:
VLC does not find a demuxer for the video stream
https://nrk-od2.telenorcdn.net/open/ps/dnpr/dnpr63701115/d0bc193d-1.smil/sc-gaFEAQ/v4_index.m3u8
or for the audio stream
https://nrk-od2.telenorcdn.net/open/ps/dnpr/dnpr63701115/d0bc193d-1.smil/sc-gaFEAQ/a0_index.m3u8
OMXplayer does play both of them.
Raspberry 3B+, Bullseye light, VLC 3.0.20 Vetinari

jc-kynesim, 2024-01-08:
Not so much demuxer fail as failure to find the 1st segment of the HLS stream - not sure why but I'll look at it.

jc-kynesim, 2024-01-08:
I have a fix (that site doesn't like the port number on the Host line which I've got to assume others do) - it'll be in the next update.

jc-kynesim, 2024-01-08:
After a bit more looking at standards, whilst I have a workaround it looks like it is the server that is non-compliant. I'll still build a workaround but it isn't VLCs fault.

eddsor, 2024-01-08:
Several of NRK's servers have  the same problem, but not all of them. I look forward to a solution.