RPi-Distro/vlc #102: Rpi5 and waveshare display "only" (no HDMI)

pwallner, 2024-01-08:
Maybe someone here can help, I asked [here](https://forum.videolan.org/viewtopic.php?f=13&t=163836&p=542096#p542096), but I fear it's a problem with the RPi-Distro.
I have a Rpi5 and acc. [this](https://forums.raspberrypi.com/viewtopic.php?p=2171303#p2171303) help, I have my waveshare display running with `config.txt`
```
# waveshare 1.3in - 240x240pixel - SKU19650
dtoverlay=mipi-dbi-spi,speed=32000000
dtparam=compatible=wavesku19650\0panel-mipi-dbi-spi
dtparam=write-only,cpha,cpol
dtparam=width=240,height=240,width-mm=23,height-mm=23
dtparam=reset-gpio=27,dc-gpio=25,backlight-gpio=18

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,nohdmi
max_framebuffers=2
```
with `mpv` everything is running (as well as with emulationstation):
```
root@rpi5:/opt/videos#  mpv --drm-connector=help
Available connectors for card 0 (/dev/dri/card2):
  SPI-1 (connected)

Available connectors for card 1 (/dev/dri/card1):

Available connectors for card 2 (/dev/dri/card0):
Cannot retrieve DRM resources: Operation not supported
```
```
root@rpi5:/opt/videos# mpv --no-audio  simpsons.mp4
 (+) Video --vid=1 (*) (h264 640x480 25.000fps)
     Audio --aid=1 --alang=deu (*) (aac 1ch 48000Hz)
[vo/gpu] VT_GETMODE failed: Inappropriate ioctl for device
[vo/gpu/opengl] Failed to set up VT switcher. Terminal switching will be unavailable.
VO: [gpu] 640x480 yuv420p
V: 00:05:28 / 00:22:15 (25%)
```
with `drmdevice` I get this output:
```
root@rpi5:/opt# drmdevice
--- Checking the number of DRM device available ---
--- Devices reported 3 ---
--- Retrieving devices information (PCI device revision is ignored) ---
device[0]
+-> available_nodes 0x01
+-> nodes
|   +-> nodes[0] /dev/dri/card2
+-> bustype 0002
|   +-> platform
|       +-> fullname    /axi/pcie@120000/rp1/spi@50000/panel@0
+-> deviceinfo
    +-> platform
        +-> compatible
                    wavesku19650
                    panel-mipi-dbi-spi

--- Opening device node /dev/dri/card2 ---
--- Retrieving device info, for node /dev/dri/card2 ---
device[0]
+-> available_nodes 0x01
+-> nodes
|   +-> nodes[0] /dev/dri/card2
+-> bustype 0002
|   +-> platform
|       +-> fullname    /axi/pcie@120000/rp1/spi@50000/panel@0
+-> deviceinfo
    +-> platform
        +-> compatible
                    wavesku19650
                    panel-mipi-dbi-spi

device[1]
+-> available_nodes 0x01
[... 500 lines cut ...]
[00007ffeb00013f0] drm_vout vout display error: Failed to find output <auto>: No such file or directory
[00007ffeb00013f0] drm_vout vout display debug: <<< CloseDrmVout
[00007ffeb8069b10] main spu text debug: removing module "freetype"
[00007ffeb8079860] main scale debug: removing module "yuvp"
[00007ffeb80c3cd0] main scale debug: removing module "swscale"
f^C[00005555e2511560] main libvlc debug: exiting
[00005555e2511560] main libvlc debug: removing all interfaces
[00005555e2807220] main interface debug: removing module "dummy"
[00005555e27d1730] main interface debug: removing module "hotkeys"
[00005555e27e0630] main interface debug: removing module "dbus"
[00005555e27b5b50] main playlist debug: destroying
[00005555e27b5b50] main playlist debug: deactivating the playlist
[00005555e27cf280] main audio output debug: removing module "alsa"
[00005555e27e16e0] main playlist export debug: saving media library to file /root/.local/share/vlc/ml.xspf.tmp1503
[00005555e27e16e0] main playlist export debug: looking for playlist export module matching "export-xspf": 4 candidates
[00005555e27e16e0] main playlist export debug: using playlist export module "export"
[00005555e27e16e0] main playlist export debug: removing module "export"
[00005555e27b5b50] main playlist debug: deleting item `Media Library'
[00005555e27b5b50] main playlist debug: deleting item `simpsons.mp4'
[00005555e27b5b50] main playlist debug: deleting item `Playlist'
[00005555e2798ad0] main keystore debug: removing module "memory"
```

It doesn't matter which parameter I tried, I never get the video shown on the display...
Please help!!!

jc-kynesim, 2024-01-08:
Are you trying this from outside any sort of windowing environment just from the command line so you would hope that DRM output worked? If so try `vlc -V drm-vout --drm-vout-display SPI-1 ...` That may/may not work - The Pi VLC DRM output expects a moderately capable output and from the debug above I'm not sure it has everything it is expecting. Either way give it a go and see if we've got further.

pwallner, 2024-01-09:
@jc-kynesim Yes I booted the rpi to cli with autologin (not to desktop). Then I started with SSH session with vlc 3.0.20 (installed with apt)
```
root@rpi5:/opt/videos# cvlc --no-audio -V drm_vout --drm-vout-display SPI-1 simpsons.mp4
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[000055557f57a2d0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[000055557f2dc790] main interface error: no suitable interface module
[000055557f2b9560] main libvlc error: interface "globalhotkeys,none" initialization failed
[000055557f2dc790] dummy interface: using the dummy interface module...
[h264_v4l2m2m @ 0x7ffebcc12ff0] Could not find a valid device
[h264_v4l2m2m @ 0x7ffebcc12ff0] can't configure decoder
[00007ffebcc080a0] avcodec decoder error: cannot start codec (h264_v4l2m2m)
[00007ffeb40013f0] drm_vout vout display: <<< OpenDrmVout: Fmt=I420
[00007ffeb40013f0] drm_vout vout display error: Failed to get xlease
[00007ffeb40013f0] drm_vout vout display error: Failed to find output SPI-1: No such file or directory
[00007ffeb0066020] main video output error: video output creation failed
[00007ffebcc080a0] main decoder error: failed to create video output
[h264 @ 0x7ffebcc19160] get_buffer() failed
[h264 @ 0x7ffebcc19160] thread_get_buffer() failed
[h264 @ 0x7ffebcc19160] decode_slice_header error
[h264 @ 0x7ffebcc19160] no frame!
```
and as discussed in parallel [here](https://forum.videolan.org/viewtopic.php?f=13&t=163836&p=542109#) with compiled version 4.0.0 with this error:
```
root@rpi5:/opt/videos# cvlc --kms-device=/dev/dri/card2 --no-audio simpsons.mp4 
VLC media player 4.0.0-dev Otto Chriek (revision 4.0.0-dev-27260-g3b8438caad)
[00005555f987dc60] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[00005555f98aff20] main interface error: no suitable interface module
[00005555f98acf70] dummy interface: using the dummy interface module...
[00007ffec8099c10] kms window: Updating list of connectors:
[00007ffec8099c10] kms window: Looping over 1 resources
[00007ffec8099c10] kms window: connector 0: SPI-1
[00007ffef0ed53b8] xcb generic error: window not available
[00007ffef0ed5368] xcb generic error: window not available
[00007ffec80f0f60] drm_display vout display error: DRM plane format error: Operation not supported
[00007ffec8046580] main video output error: video output display creation failed
[h264 @ 0x7ffedc218910] get_buffer() failed
[h264 @ 0x7ffedc218910] thread_get_buffer() failed
[h264 @ 0x7ffedc218910] decode_slice_header error
[h264 @ 0x7ffedc218910] no frame!
[00007ffec8099c10] kms window: Looping over 1 resources
[00007ffec8099c10] kms window: connector 0: SPI-1
[00007ffed60953b8] xcb generic error: window not available
[00007ffed6095368] xcb generic error: window not available
[00007ffec4048950] drm_display vout display error: DRM plane format error: Operation not supported
[00007ffec8046580] main video output error: video output display creation failed
[h264 @ 0x7ffedc3115e0] get_buffer() failed
[h264 @ 0x7ffedc3115e0] thread_get_buffer() failed
[h264 @ 0x7ffedc3115e0] decode_slice_header error
[h264 @ 0x7ffedc3115e0] no frame!
```

jc-kynesim, 2024-01-09:
Thanks for that - I'm surprised that v3 failed to open SPI-1, I'll see if I can work out why - I'd have expected that much to work.

pwallner, 2024-01-09:
@jc-kynesim with v4 see [here](https://forum.videolan.org/viewtopic.php?f=13&t=163836&p=542109#p542113) they say it's a driver bug?! I wonder, because `mpv` is working. So maybe a patch at `mipi-dbi-spi` necessary?

jc-kynesim, 2024-01-09:
Maybe, but I suspect it is simply that the limited formats that your display supports don't match what VLC wants. Note this is just a guess - I haven't looked into it and they might be right. I've now picked up an SPI display so I can debug the v3 code which is indeed broken when trying to select a device on anything not attached to the Pis HVS.

pwallner, 2024-01-10:
@jc-kynesim "Fun" fact. Everything above is when I try in a ssh session, I attached a keyboard and started clvc and the video is shown on the display:
[image]
I used all for me known parameters, but cvlc doesn't respect the aspect ratio..

When I start mpv without any paramter with the attached keyboard, the aspect ratio is ok (mpv also works via ssh session!):
[image]

jc-kynesim, 2024-01-11:
Was the "working" cvlc v4 or v3 (or both)?
I'm not as surprised as I might be as I've now got a SPI display and they are a LOT stupider than the HVS-driven HDMI outputs. In particular they have different behaviour to HVS if you try to set stuff without a plane attached to the output. Having a command line is probably sufficient to init the display state such that VLC can get at it.
Likewise the scaling - the only option is to put the entire plane on the display with no borders (HVS is happy to put black everywhere you have no video) so if you want borders they need to be constructed manually.

pwallner, 2024-01-11:
@jc-kynesim It was VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b) from apt source. Didn't test v4 for now because v3 was working.

What I don't unterstand, why the simple mpv has no problems... It works from cli and ssh and has no problems with aspect ratio.

pwallner, 2024-02-02:
> The key word there is "simple". My code (at least) was written to take advantage of what the HVS could do for me (i.e. enable 4k video output) and so uses features that (a) make life simpler / faster / without frame copy or conversion and (b) don't exist on your device. I'll fix that.

Don't want to push you, but did you find something in the source for fixing?

jc-kynesim, 2024-02-05:
Yeah - next release (shortly) should be better. I don't guarantee it'll work for you but I've got a couple of dumb displays that work for me now

mcpat-it, 2024-03-01:
@jc-kynesim  could it be you did it? I just updated vlc and it's running now :heart: Means, I started from ssh session, video is playing on small display and correct ratio!! :heart: