RPi-Distro/vlc #107: VLC player adds black screen during looping

fctr, 2024-05-07:
One of the nicest features of OMXPlayer was that it didn't "hiccup" when playing the next video. It was seamless, so looping had no black screens or blanking. VLC (especially using framebuffer) does this and it's really, really bad for things like video billboards. I've tried all kinds of tricks but to no avail.

tofuSCHNITZEL, 2025-03-16:
I'm currently also playing around with vlc to replace omxplayer in https://github.com/adafruit/pi_video_looper/
its very frustrating... on the first start of the video there is like 1-3 sec black screen. the loop like you said is not seemless though I don't get a black screen in between when using -loop but there is stuttering...
I'm also looking into mpv but it has its own downsides... no hardware accelerated playback on rpi (or i did not manage to enable it) but less stutter in between videos and also way less black screen before a video starts the first time...

Madvfr, 2025-07-28:
I've been scouring for a solution to this, but have also just noticed...for some reason when the video loops the "full screen" button on the interface flashes...

Fiddling with everything fullscreeny has done nought, and neither forums nor our overlord, Das Google, has helped me . So still an issue?