RPi-Distro/vlc #109: VLC window disappear when attempting to play h265

atimmernans, 2024-08-09:
I am using 3.0.21 on a PI4 and attempting to play h265 causes the VLC window to disappear.
Playing with the options, disabling hardware acceleration of ffmpeg allows me to play (barely) to play such videos.

I did not have to do anything in the past to play h265 and I kinda remember there were some recent updates of libraries related to video. So is the library now attempting to use hardware acceleration for h265 but the PI4 doesn't support it or is that support broken? Checking in preferences some ffmpeg 8:4.3.7+deb11u1+rpt1 are installed (a pity it doesn't mention when it was installed/updated)

popcornmix, 2024-08-09:
Are you on RPiOS Bookworm 64-bit (or something else)?
How are you starting video? Just the Media->open file menu option?

Can you confirm if it happens with a file we can also test. e.g. [this](https://test-videos.co.uk/vids/bigbuckbunny/mp4/h265/720/Big_Buck_Bunny_720_10s_30MB.mp4).

atimmernans, 2024-08-09:
I am on RPiOS Bullseye 64-bit.
It doesn't matter how I open the file Media->open file or double-click. Though for that later I have the hourglass spinning for a while.
The test file triggers the problem.

jc-kynesim, 2024-08-13:
I don't think that we've released a 3.0.21 for Bullseye. If you've just
grabbed a Bookworm image I'm going to guess that at least some of the
dependencies are bust. Do you have a pressing reason for using Bullseye
rather than Bookworm?

XECDesign, 2024-08-13:
Just had a look and I did ship vlc 1:3.0.21-0+rpt1+deb11u1 a while ago. It's using the same patches as they previous version of the bullseye package, since they applied cleanly.

atimmernans, 2024-08-24:
Fixed by the new release of ffmpeg and associated libraries.

closed 2024-08-24 by atimmernans