RPi-Distro/vlc #110: VLC 1:3.0.21-0+rpt2+deb12u1 doesn't work with cage compositor

ykharuzhy, 2024-09-27:
Latest  1:3.0.21-0+rpt2+deb12u1 VLC version doesn't show any video when running under cage (minimalistic kiosk-mode compositor).

1:3.0.21-0+rpt1+deb12u1 works (btw, the latest sources look like were not pushed here).

Cage project page: https://github.com/cage-kiosk/cage

Log: [cvlc-cage.log](https://github.com/user-attachments/files/17166032/cvlc-cage.log)

Run command: cage -- cvlc --verbose=2 ~/media/data/IMG_1625.mov

To reproduce, disable any desktop manager, install cage and seatd and try to run above command.

XECDesign, 2024-09-27:
I don't believe we support compositors other than wayfire and labwc. And going forward, it's going to be only labwc. However, if you have a fix, I'm happy to merge it.

jc-kynesim, 2024-09-27:
Life will likely get better with the next released VLC. That particular command triggered an early version of wl_xdg_shell that I wrote to help with testing. The next version will have a lot more robust support (I've just tried it here).
It would be nice if cage supported wp_viewporter (needed by wl_dmabuf), but we seem to fall back to GL OK.

jc-kynesim, 2024-09-27:
Why do you want cage anyway?

ykharuzhy, 2024-09-30:
Latest cage has viewporter support, btw.

I am preparing distro for RPi-based media player and need to run VLC as 'kiosk' application: fullscreen mode without of any decorations and input interactions (DRM output is not an option because lack of rotation support). Cage was the simplest way to achieve this found by google :) If there is any alternative with similar simple configuration, i will glad to test it.

jc-kynesim, 2024-09-30:
If you need 90degree rotation support then yes GL is going to be your best bet currently. There's going to be a new VLC quite soon - I'm just trying to kill some leaks that have turned up.