RPi-Distro/vlc #111: Crashes when 2x scale on wayfire

edgerunning, 2024-10-12:
I am using the default desktop environment of Raspberry Pi OS Bookworm.  I set the scale factor with configuration like 
```
[output:HDMI-A-1]
mode = 3840x2160@60000
position = 0,0
transform = normal
scale = 2.0
```
in the ~/.config/wayfire.ini file
with this setup, vlc can't play videos and just crashes.

jc-kynesim, 2024-10-22:
Yup - reproduced on both wayfire and labwc. This appears to be a wayland/wl-roots/... bug. The black surface that is used for the background is a single-pixel-buffer (special wayland single colour buffer) and wayland complains that 1x1 pixels isn't a multiple of 2! I'll see if I can trace this a bit deeper and report it back to the wayland lists.
You can somwhat work around this by adding `--wl-dmabuf-chequerboard` to the VLC command-line which changes the background to a chequerboard for testing. You then get video. It is the wrong size -  it is twice the expected size and spills over the confines of the VLC window. This is a VLC issue and I'll see if I can track it down.

jc-kynesim, 2024-10-22:
A better workaround would be to use `-V opengles2` which forces the EGL render path that does get this right and doesn't use single pixel buffers but is potentially a little less efficient than the wl-dmabuf path.