RPi-Distro/vlc #112: Multichannel PCM or AAC only playing as stereo over HDMI

herrernst, 2024-11-06:
I have some media files which have multichannel audio (5.1 or 7.1) as PCM or AAC. I'm not able to play them on my AVR with VLC as multichannel, the AVR only receives stereo.

You can test with 7.1 PCM wave file: https://www2.iis.fraunhofer.de/AAC/7.1auditionOutLeader%20v2.wav

I've tried all of this (on virtual console, no X11 or Wayland):
```
vlc "7.1auditionOutLeader v2.wav"
vlc --aout alsa:hw "7.1auditionOutLeader v2.wav"
vlc --aout alsa:hdmi "7.1auditionOutLeader v2.wav"
vlc --aout alsa:hdmi --spdif --alsa-passthrough 2 "7.1auditionOutLeader v2.wav"
vlc --aout alsa:hdmi --spdif "7.1auditionOutLeader v2.wav"
vlc --aout alsa:hdmi --spdif --alsa-passthrough 1 "7.1auditionOutLeader v2.wav"
```
AVR only gets stereo. Same with 5.1 AAC from https://archive.org/download/5.1SurroundSoundTestFilesVariousFormatsAACAC3MP4DTSWAV/5.1%20Surround%20Sound%20AAC%20Test.mp4

But using `aplay "7.1auditionOutLeader v2.wav"` and `speaker-test -c 8` works (though with wrong channel mapping, but that's another bug)

Passthrough of DTS, AC3 e.g. also work fine though with `vlc --aout alsa:hdmi --spdif --alsa-passthrough 2 https://forum.videohelp.com/attachments/29969-1422663834/11sec.ac3`

This is on Raspberry Pi OS 12, VLC media player 3.0.21 Vetinari (revision 3.0.21-0-gdd8bfdbabe8)

Testing with the same files on Raspberry Pi OS 10 and OMXPlayer, I get them to work with `omxplayer -p "7.1auditionOutLeader v2.wav"` (although it's weird that `-p` for passthrough is necessary, and also channel order is wrong).

jc-kynesim, 2024-11-25:
Sorry about the delay. Try

`vlc -A alsa --alsa-audio-device hdmi --alsa-audio-channels "Surround 7.1" -vv 7.1auditionOutLeader\ v2.wav`

The "-vv" just adds (lots of) debug to send back here if that doesn't work. Remove for actual use. The VLC ALSA output downmixes everything that isn't passthrough to stereo unless given other options - this isn't a Pi thing its a VLC thing. I'm not completely convinced by it but I can see cases where ALSA can see more outputs than are actually connected and stereo is always safe. There's also a mapping stage that may be confused. If you end up with stereo or something other than what you were after please install and run edid=decode with

```
sudo apt install edid-decode
edid-decode /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
```
The path may require tweaking depending on platform but that worked on my Pi4. Somewhere in its output you should hopefully find a "Speaker Allocation Data Block". Paste that in your reply.

herrernst, 2024-11-25:
No need to be sorry! Thanks you for your answer, better late than never.

Below is the output of `vlc -A alsa --alsa-audio-device hdmi --alsa-audio-channels "Surround 7.1" -vv 7.1auditionOutLeader\ v2.wav` (it also only plays on 2 channels):
```
VLC media player 3.0.21 Vetinari
Command Line Interface initialized. Type `help' for help.
> Shutting down.
[00000055784cdce0] main libvlc debug: VLC media player - 3.0.21 Vetinari
[00000055784cdce0] main libvlc debug: Copyright ¬© 1996-2024 the VideoLAN team
[00000055784cdce0] main libvlc debug: revision 3.0.21-0-gdd8bfdbabe8
[00000055784cdce0] main libvlc debug: configured with ./configure  '--build=aarch64-linux-gnu' '--prefix=/usr' '--includedir=${prefix}/include' '--mandir=${prefix}/share/man' '--infodir=${prefix}/share/info' '--sysconfdir=/etc' '--localstatedir=/var' '--disable-option-checking' '--disable-silent-rules' '--libdir=${prefix}/lib/aarch64-linux-gnu' '--runstatedir=/run' '--disable-maintainer-mode' '--disable-dependency-tracking' '--disable-debug' '--config-cache' '--disable-update-check' '--enable-fast-install' '--docdir=/usr/share/doc/vlc' '--with-binary-version=1:3.0.21-0+rpt2+deb12u1' '--enable-a52' '--enable-aa' '--enable-aribsub' '--enable-avahi' '--enable-bluray' '--enable-caca' '--enable-chromaprint' '--enable-chromecast' '--enable-dav1d' '--enable-dbus' '--enable-dca' '--enable-dvbpsi' '--enable-dvdnav' '--enable-faad' '--enable-flac' '--enable-fluidsynth' '--enable-freetype' '--enable-fribidi' '--enable-gles2' '--enable-gnutls' '--enable-harfbuzz' '--enable-jack' '--enable-kate' '--enable-libass' '--enable-libmpeg2' '--enable-libxml2' '--enable-lirc' '--enable-live555' '--enable-mad' '--enable-matroska' '--enable-mod' '--enable-mpc' '--enable-mpg123' '--enable-mtp' '--enable-ncurses' '--enable-notify' '--enable-ogg' '--enable-opus' '--enable-pulse' '--enable-qt' '--enable-realrtsp' '--enable-samplerate' '--enable-sdl-image' '--enable-sftp' '--enable-shine' '--enable-shout' '--enable-skins2' '--enable-soxr' '--enable-spatialaudio' '--enable-speex' '--enable-srt' '--enable-svg' '--enable-svgdec' '--enable-taglib' '--enable-theora' '--enable-twolame' '--enable-upnp' '--enable-vnc' '--enable-vorbis' '--enable-x264' '--enable-x265' '--enable-zvbi' '--with-kde-solid=/usr/share/solid/actions/' '--disable-aom' '--disable-crystalhd' '--disable-d3d11va' '--disable-decklink' '--disable-directx' '--disable-dsm' '--disable-dxva2' '--disable-fdkaac' '--disable-fluidlite' '--disable-freerdp' '--disable-goom' '--disable-gst-decode' '--disable-libtar' '--disable-macosx' '--disable-macosx-avfoundation' '--disable-macosx-qtkit' '--disable-mfx' '--disable-microdns' '--disable-mmal' '--disable-mmal-avcodec' '--disable-opencv' '--disable-projectm' '--disable-schroedinger' '--disable-sndio' '--disable-sparkle' '--disable-telx' '--disable-vpx' '--disable-vsxu' '--disable-wasapi' '--enable-alsa' '--enable-dc1394' '--enable-dv1394' '--enable-libplacebo' '--enable-linsys' '--enable-nfs' '--enable-udev' '--enable-v4l2' '--enable-wayland' '--enable-vcd' '--enable-smbclient' '--disable-oss' '--disable-mmx' '--disable-sse' '--disable-neon' '--disable-vdpau' '--disable-libva' '--disable-altivec' '--disable-omxil' 'build_alias=aarch64-linux-gnu' 'CFLAGS=-g -O2 -ffile-prefix-map=/build/vlc-Yx7fPf/vlc-3.0.21=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -mbranch-protection=standard ' 'LDFLAGS=-Wl,-z,relro -Wl,-z,now' 'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' 'CXXFLAGS=-g -O2 -ffile-prefix-map=/build/vlc-Yx7fPf/vlc-3.0.21=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -mbranch-protection=standard ' 'OBJCFLAGS=-g -O2 -ffile-prefix-map=/build/vlc-Yx7fPf/vlc-3.0.21=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -mbranch-protection=standard'
[00000055784cdce0] main libvlc debug: searching plug-in modules
[00000055784cdce0] main libvlc debug: loading plugins cache file /usr/lib/aarch64-linux-gnu/vlc/plugins/plugins.dat
[00000055784cdce0] main libvlc debug: recursively browsing `/usr/lib/aarch64-linux-gnu/vlc/plugins'
[00000055784cdce0] main libvlc debug: plug-ins loaded: 490 modules
[00000055784cdce0] main libvlc debug: opening config file (/home/user/.config/vlc/vlcrc)
[00000055784ce040] main logger debug: looking for logger module matching "any": 4 candidates
[00000055784ce040] main logger debug: using logger module "console"
[00000055784cdce0] main libvlc debug: translation test: code is "C"
[00000055785618f0] main keystore debug: looking for keystore module matching "memory": 4 candidates
[00000055785618f0] main keystore debug: using keystore module "memory"
[00000055784cdce0] main libvlc debug: CPU has capabilities FPU 
[0000005578567ba0] main input debug: Creating an input for 'Media Library'
[0000005578567ba0] main input debug: Input is a meta file: disabling unneeded options
[0000005578567ba0] main input debug: using timeshift granularity of 50 MiB
[0000005578567ba0] main input debug: using default timeshift path
[0000005578567ba0] main input debug: `file/directory:///home/user/.local/share/vlc/ml.xspf' gives access `file' demux `directory' path `/home/user/.local/share/vlc/ml.xspf'
[0000005578561a90] main input source debug: creating demux: access='file' demux='directory' location='/home/user/.local/share/vlc/ml.xspf' file='/home/user/.local/share/vlc/ml.xspf'
[... cut ...]
[0000007f8800eb80] main playlist export debug: looking for playlist export module matching "export-xspf": 4 candidates
[0000007f8800eb80] main playlist export debug: using playlist export module "export"
[0000007f8800eb80] main playlist export debug: removing module "export"
[0000005578562b60] main playlist debug: deleting item `Media Library'
[0000005578562b60] main playlist debug: deleting item `7.1auditionOutLeader v2.wav'
[0000005578562b60] main playlist debug: deleting item `Playlist'
[00000055785618f0] main keystore debug: removing module "memory"
```

herrernst, 2024-11-25:
The speaker allocation data block that my receiver reports is unfortunately not correct:
```
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
    LFE1 - Low Frequency Effects 1
    FC - Front Center
    BL/BR - Back Left/Right
    BC - Back Center
    RLC/RRC - Rear Left/Right of Center (Deprecated)
```
whereas I have 5.1.2 (i. e. 5.1 and two middle top for Atmos)

herrernst, 2024-11-25:
Sorry for splitting up my answers ... I've now edited the EDID for a conventional 5.1 layout with an EDID editor, and loaded it with `drm.edid_firmware=` kernel command line paramter.

`edid-decode` now correctly reports:
```
  Speaker Allocation Data Block:
    FL/FR - Front Left/Right
    LFE1 - Low Frequency Effects 1
    FC - Front Center
    LS/RS - Left/Right Surround
```
But VLC keeps playing in stereo, and `speaker-test -c 8` also is bogus:
```
 0 - Front Left
 3 - Front Center
 1 - Front Right
 2 - LFE
 4 - Unknown
 5 - Unknown
 6 - Unknown
 7 - Unknown
```

jc-kynesim, 2024-11-26:
Huh! Thats annoying. My bluesound bar reports the same dodgy edid as yours and VLC ends up falling back to 5.1 (dropping RLC/RRC). What's your setup? Is this a desktop install or a lite? Are you trying this in the desktop or on the command line?

I clearly need to do more testing, but FYI I was testing with a desktop install (so this will all have been going via pipewire) booted to the command line so ALSA will probably have had exclusive use of the output.

Thanks for the info - I think the next version of Pi VLC will have to have some more useful debug in its ALSA path.

jc-kynesim, 2024-11-26:
As an aside given that I now have two examples posting EDIDs with RLC/RRC I'm going to guess that that was a standard mistake - if those should have been middle channels than I might add an explicit kludge for the setup (or maybe RL/RR were widdle and RRC/RLC were rear?)

herrernst, 2024-11-26:
> Huh! Thats annoying. My bluesound bar reports the same dodgy edid as yours and VLC ends up falling back to 5.1 (dropping RLC/RRC). 

So at least 5.1 works for you?!

> What's your setup? Is this a desktop install or a lite? Are you trying this in the desktop or on the command line?

It's a lite install and I'm using command line (so direct also, no Pipewire).

herrernst, 2024-11-26:
> As an aside given that I now have two examples posting EDIDs with RLC/RRC I'm going to guess that that was a standard mistake - if those should have been middle channels than I might add an explicit kludge for the setup (or maybe RL/RR were widdle and RRC/RLC were rear?)

Sorry, I don't quite understand 🙈 (Myself, I have only 5.1, so "side" surrounds ("Ls", "Rs"), no rears.)

jc-kynesim, 2024-11-26:
Sorry - didn't read your post correctly - I completely missed the bit where you said 5.1.2. I assumed that given the file you gave me was 7.1 you had 7.1. I must read the entire answer next time...
It turns out that of the 3 setups I've now looked at all report your edid.

Yes I get downmix to 5.1 on my setup - pipewire amazingly seems to make both pulse & alsa work better. I'll need to do some more testing to see what lite does / doesn't do.

I _think_ I now understand the deprecated RRC/RLC settings. From what I've read they are an alternative to a single RC channel often containing the same data and whilst both are declared they aren't expected to be used at the same time. The way channels are assigned has varied between earlier and later standards and is confusing.

I'm going to write a trivial utility to dump the channel maps that ALSA reports. I'm sure something should already do this but I haven't found it. Once I've done that it'll be easier to see what VLC should be doing given the info it has.

jc-kynesim, 2024-11-26:
I've written a trivial alsa map report. The source is in https://github.com/jc-kynesim/alsamap.git can you build and run that (./alsamap hdmi) please? If you can't build I can provide a binary. If it doesn't build or explodes when you run it please file a bug against it!
It should list all the channel mappings that a given device supports.
Thanks

herrernst, 2024-11-26:
Thank you again! Here is the output of `alsamap` (it's the same whether using the default EDID of the AVR or a custom override):
```
[ 0] Fixed: Chans=2: FL(3) FR(4)
[ 1] Fixed: Chans=4: FL(3) FR(4) LFE(8) NA(1)
[ 2] Fixed: Chans=4: FL(3) FR(4) NA(1) FC(7)
[ 3] Fixed: Chans=4: FL(3) FR(4) LFE(8) FC(7)
[ 4] Fixed: Chans=6: FL(3) FR(4) NA(1) NA(1) RC(11) NA(1)
[ 5] Fixed: Chans=6: FL(3) FR(4) LFE(8) NA(1) RC(11) NA(1)
[ 6] Fixed: Chans=6: FL(3) FR(4) NA(1) FC(7) RC(11) NA(1)
[ 7] Fixed: Chans=6: FL(3) FR(4) LFE(8) FC(7) RC(11) NA(1)
[ 8] Fixed: Chans=6: FL(3) FR(4) NA(1) NA(1) RL(5) RR(6)
[ 9] Fixed: Chans=6: FL(3) FR(4) LFE(8) NA(1) RL(5) RR(6)
[10] Fixed: Chans=6: FL(3) FR(4) NA(1) FC(7) RL(5) RR(6)
[11] Fixed: Chans=6: FL(3) FR(4) LFE(8) FC(7) RL(5) RR(6)
[12] Fixed: Chans=8: FL(3) FR(4) NA(1) NA(1) RL(5) RR(6) RC(11) NA(1)
[13] Fixed: Chans=8: FL(3) FR(4) LFE(8) NA(1) RL(5) RR(6) RC(11) NA(1)
[14] Fixed: Chans=8: FL(3) FR(4) NA(1) FC(7) RL(5) RR(6) RC(11) NA(1)
[15] Fixed: Chans=8: FL(3) FR(4) LFE(8) FC(7) RL(5) RR(6) RC(11) NA(1)
[16] Fixed: Chans=8: FL(3) FR(4) NA(1) NA(1) RL(5) RR(6) RLC(14) RRC(15)
[17] Fixed: Chans=8: FL(3) FR(4) LFE(8) NA(1) RL(5) RR(6) RLC(14) RRC(15)
[18] Fixed: Chans=8: FL(3) FR(4) NA(1) FC(7) RL(5) RR(6) RLC(14) RRC(15)
[19] Fixed: Chans=8: FL(3) FR(4) LFE(8) FC(7) RL(5) RR(6) RLC(14) RRC(15)
[20] Fixed: Chans=8: FL(3) FR(4) NA(1) NA(1) NA(1) NA(1) FLC(12) FRC(13)
[21] Fixed: Chans=8: FL(3) FR(4) LFE(8) NA(1) NA(1) NA(1) FLC(12) FRC(13)
[22] Fixed: Chans=8: FL(3) FR(4) NA(1) FC(7) NA(1) NA(1) FLC(12) FRC(13)
[23] Fixed: Chans=8: FL(3) FR(4) LFE(8) FC(7) NA(1) NA(1) FLC(12) FRC(13)
[24] Fixed: Chans=8: FL(3) FR(4) NA(1) NA(1) NA(1) NA(1) FLC(12) FRC(13)
[25] Fixed: Chans=8: FL(3) FR(4) LFE(8) NA(1) NA(1) NA(1) FLC(12) FRC(13)
[26] Fixed: Chans=8: FL(3) FR(4) NA(1) FC(7) NA(1) NA(1) FLC(12) FRC(13)
[27] Fixed: Chans=8: FL(3) FR(4) LFE(8) FC(7) NA(1) NA(1) FLC(12) FRC(13)
[28] Fixed: Chans=8: FL(3) FR(4) NA(1) NA(1) NA(1) NA(1) FLC(12) FRC(13)
[29] Fixed: Chans=8: FL(3) FR(4) LFE(8) NA(1) NA(1) NA(1) FLC(12) FRC(13)
[30] Fixed: Chans=8: FL(3) FR(4) NA(1) FC(7) NA(1) NA(1) FLC(12) FRC(13)
[31] Fixed: Chans=8: FL(3) FR(4) LFE(8) FC(7) NA(1) NA(1) FLC(12) FRC(13)
```

jc-kynesim, 2024-11-26:
OK - not sure how my code at home was working - I must have had a bit of debug code left it that enabled life - try this: `vlc -A alsa --alsa-audio-device hdmi --alsa-audio-channels 4976 -vv 7.1auditionOutLeader\ v2.wav ` I thought that VLC converted text to numbers when getting that option but it appears that is only a hint for help :-( 4967 gives the channel mask for 7.1 which I expect to fail but you should fall back to 5.1. 4199 should ask for 5.1. The values are in the help from vlc -H (you will want to pipe the output from this to a file and search it).

herrernst, 2024-11-26:
Thank you so much! Using `--alsa-audio-channels` works. I must apologize for not having found this earlier, but I would have expected VLC to auto-detect available channels. Thanks again!

closed 2024-11-26 by herrernst

jc-kynesim, 2024-11-27:
I should have been able to give you the answer in my first reply. At least I know a lot more about ALSA & HDMI audio now.

herrernst, 2024-11-29:
BTW if someone is interested, VLC's channel maps are defined here: https://github.com/RPi-Distro/vlc/blob/bookworm-rpt/include/vlc_es.h#L117