RPi-Distro/vlc #114: VLC control the whole screen if it was on full screen mode

salmahassan1626, 2025-07-06:
Hello everyone,

I'm using VLC on a Raspberry Pi Compute Module 4 (CM4), running it from within a Qt application in fullscreen mode on X11. The problem is that when VLC enters fullscreen mode, it seems to take over the entire display, and other applications running on the same screen are not visible — they render in the background.

Use Case:
I'm building a media solution that includes:

A Qt-based app that uses VLC for video playback in fullscreen mode.

A secondary Qt app (such as a volume OSD indicator) that should appear on top of the video when triggered (e.g., using an RCU or keyboard input).

Currently, although the volume is changed successfully in the backend, the OSD is not shown on the screen because VLC is consuming or blocking the entire display layer.

Expected Behavior:
Even in fullscreen mode, VLC should allow other X11 windows to appear above it (e.g., dialogs, OSDs, etc.), just like typical fullscreen apps on Linux desktops.

Actual Behavior:
Other apps are rendered in the background, and VLC fully covers the screen, making it impossible to overlay additional UI components.

jc-kynesim, 2025-07-07:
Yup - that is intentional. Under X the default video out uses DRM for fullscreen display and to do that it leases the DRM handles that X uses thus preventing X for displaying anything. It does this because DRM output is much more efficient than anything else and is the only way you are going to get smooth 4k or 1080p60 output. If you need windows then add a "-V opengles2" to your command line and that will disable the fullscreen switching.

jc-kynesim, 2025-07-14:
I don't think you can easily do what you want with DRM as the output - whilst DRM itself could do it there are issues which come down to DRM being a single master interface and very tricky to share around between processes (and impossible if they aren't cooperating). X will definitely let you do what you want but will be "slower" than DRM. Your best bet is a current Wayland, which when running under labwc (i.e. the PiOS default) will do the fullscreen mode you want and allow windows on top. If the number of surfaces being displayed is small (the exact definition of small I'm not sure about - might be 2 but I think its got a bit bigger recently) then you should get DRM passthrough so it should be nearly as efficient as pure DRM, and if it can't do DRM passthrough it will fallback seamlessly to GPU compositing.