RPi-Distro/vlc #115: First few seconds of video always cuts in Rpi OS Lite

VC8597, 2025-07-25:
Raspberry pi zero 2w with raspberry pi OS 32 bit bullseye lite without desktop environment.

VLC version 3.0.21-0-gdd8bfdbabe8

Kernel - linux 6.1.21-v7+

VLC command - "cvlc --fullscreen --play-and-exit --no-osd path to video.mp4"
Video playback is problematic. Initial few frames of the video always cuts. And if I use a different HDMI display with 1024x600 pixels resolution, then more initial seconds of video gets cut.
Attaching the full vlc log by the name - vlc_nofullos_log.

[vlc_nofullos_log.txt](https://github.com/user-attachments/files/21427808/vlc_nofullos_log.txt)

jc-kynesim, 2025-07-25:
Yeah - the issue is likely to be that drm output (which is what you will be using) by default sets the output mode to match the video, and typically this takes a bit of time for the monitor/tv to catch up with during which output is black. Try adding --drm-vout-no-modeset to avoid this behaviour

VC8597, 2025-07-26:
Hey, Thank you very very much. I was stuck on this problem since last two months. Tried every possible way. 
And at last, I got to a solution of using a minimalistic desktop environment using only openbox window manager, and then running cvlc from that session. It solved the problem, but involved some steps to arrive there.

Your solution just saved me.
And also, is it okay that during a video playback, vlc uses around 20% - 26% of CPU.

closed 2025-07-28 by XECDesign