RPi-Distro/vlc #116: Particular HEVC file won't play on VLC on PRI5, but does on LibreELEC

MCucksey, 2025-08-03:
**Hardware/OS** – Raspberry PI OS (64-bit) running on a Raspberry PI 5

I’ve got a video clip that according to ffprobe is encoded as a hevc/h265, but it won’t play properly on VLC 3.0.21. It will show the first two or three frames but nothing else. The Audio is playing fine. If I seek to somewhere else in the video, again the first few frames will show, but nothing else.

I've ran plenty of other H265 files just to make sure the hardware decoder was working, and they all play fine.

The file in question is here - [OneDrive Link](https://1drv.ms/v/c/3a78f772f76f4898/ETtHJaMvXSxDvQJsYrZSRt8Bm2dZIJ02Kgsfx6pT5JG1Wg?e=4cRK4Q)

One thing I notice that's different between this file and the ones that run file, are the pixel format used. The problematic files uses yuvj420p, whereas the ones that play fine use yuv420p.

However, as mentioned in the title, I ran LibreELEC on the same hardware, and it runs fine... so sort of scratching my head at this one.

I've looked at the logging for both VLC and LibreELEC, and the only thing I can see is that at some point ffmpeg starts the hardware decoder in libreELEC, but not in VLC.

I've spent 4 days on this one and I'm not sure where to look next.

Thanks

jc-kynesim, 2025-09-08:
I'll look at that. yuv420j is just full range yuv420. Use of that ffmpeg format is now deprecated in ffmpeg (it should just be 420 with the range flag set elsewhere in the descriptor). There will be a translation failure between format values somewhere as nothing but ffmpeg thinks yuv420j is a distinct format so it need special case code.

jc-kynesim, 2025-09-08:
I dunno how you put stuff into pastebin but the 2nd "logfile" takes me to somewhere that wants me to allow notifications & tells me I have viruses. Not touching this.

jc-kynesim, 2025-09-08:
Maybe pastebin's advertising has got something bad (I get to briefly see something that look like a logfile before bad stuff happens). You have onedrive - try repasting there.

jc-kynesim, 2025-09-08:
Mental note to self - always open stuff in a sandbox.

jc-kynesim, 2025-09-08:
OK - that was better.

1st off,  what are you using as a desktop? It doesn't appear to be Wayland (labwc).
2nd, is that all the log of the failure? I'm mildly surprised we don't get anything that looks like an actual error.

If you can use wayland I'd recommend it for this purpose., its ability to throw away frames it can't render right now is superior to the EGL path. 4k windowed display is never going to be perfect on a Pi - fullscreen will be better as it allows bypassing composition.

MCucksey, 2025-09-08:
Thank you. 

For what it's worth, I did mange to get it playing with your hello_drmprime program!

jc-kynesim, 2025-09-08:
I have a fix (it is just a failure to note that J420 == YUV420 in a couple of places). It'll be in the next VLC release.

MCucksey, 2025-09-09:
I tried this on a fresh install of Raspbian with 'sudo apt build-dep vlc' but during compile it throws..

```
vlc/modules/hw/drm/drm_av_deinterlace.c:13:10: fatal error: libavfilter/buffersink.h: No such file or directory
   13 | #include <libavfilter/buffersink.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

jc-kynesim, 2025-09-09:
Dunno - worked for me just now. PiOS 64-bit desktop (not lite, not full):
Clean install onto SD card + update from 2025-05-06-raspios-bookworm-arm64.img.xz onto Pi5
Remove #s from all the deb-src lines in /etc/apt/sources.list and /etc/apt/sources.list.d/raspi.list
```
sudo apt update
sudo apt build-dep vlc
git clone https://github.com/jc-kynesim/vlc
cd vlc
git checkout test/3.0.21-trixie/rpi_6
pi-util/conf_native.sh --make
```
built cleanly for me. You did remember to uncomment the deb-src in raspi.list?